mixelpixx/KiCAD-MCP-Server #1: 3 Issues Showed after run "npm run build"

StevenZhao-Gen, 2025-04-30:
Hi,there have 3 Issues Showed after run "npm run build",below log showed the detail.
Would you please help to confirm how to correct these issues,thanks!

> kicad-mcp@1.0.0 build
> tsc

src/resources/board.ts:11:58 - error TS2307: Cannot find module '../utils/resource-helpers.js' or its corresponding type declarations.

11 import { createJsonResponse, createBinaryResponse } from '../utils/resource-helpers.js';
                                                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/server.ts:13:38 - error TS2307: Cannot find module './tools/project.js' or its corresponding type declarations.

13 import { registerProjectTools } from './tools/project.js';
                                        ~~~~~~~~~~~~~~~~~~~~

src/server.ts:16:38 - error TS2307: Cannot find module './tools/routing.js' or its corresponding type declarations.

16 import { registerRoutingTools } from './tools/routing.js';
                                        ~~~~~~~~~~~~~~~~~~~~

src/tools/index.ts:7:38 - error TS2307: Cannot find module './project.js' or its corresponding type declarations.

7 export { registerProjectTools } from './project.js';
                                       ~~~~~~~~~~~~~~

src/tools/index.ts:10:38 - error TS2307: Cannot find module './routing.js' or its corresponding type declarations.

10 export { registerRoutingTools } from './routing.js';
                                        ~~~~~~~~~~~~~~

src/tools/index.ts:13:40 - error TS2307: Cannot find module './schematic.js' or its corresponding type declarations.

13 export { registerSchematicTools } from './schematic.js';
                                          ~~~~~~~~~~~~~~~~

Found 6 errors in 3 files.

Errors  Files
     1  src/resources/board.ts:11
     2  src/server.ts:13
     3  src/tools/index.ts:7

buishuohua, 2025-05-01:
I encountered with the same errors

nargetdev, 2025-05-06:
same.

The `dist` pack is out of sync with the `src` pack .. i would assume all .js within `dist` should be derived from .ts (source of truth) in the `src` dir .. but this is not the case right now.

Also should be noted that all python paths are currently hardcoded as  C:\\  wintard paths

crisap94, 2025-05-20:
The Documentation was generated by an AI. Any luck with this installation?

ImNoahDev, 2025-05-31:
I got the one at that link working quite easily but it doesnt seem to be able to generate schematics like this one (claims) it can.

mixelpixx, 2025-12-28:
## ✅ Build Issues Resolved

Good news! The TypeScript build errors mentioned in this issue have been resolved in the current codebase.

**Tested on latest main branch (Dec 28, 2025):**
- ✅ `npm install` completes successfully
- ✅ `npm run build` compiles without errors
- ✅ All module imports resolve correctly

**What was fixed:**
- Package dependencies are now compatible
- TypeScript module resolution working correctly with ES modules
- All `.js` import extensions resolve to their `.ts` source files

**To resolve this issue:**
1. Pull the latest code from main branch
2. Run `npm install`
3. Run `npm run build`

The build should complete successfully with no errors.

**Fresh install test results:**
```
added 131 packages, and audited 132 packages in 7s
> npm run build
> tsc
✓ Build completed successfully
```

Please try the latest version and let us know if you encounter any issues!

closed 2025-12-28 by mixelpixx