mixelpixx/KiCAD-MCP-Server #3: Issues in npm run build

TS-Hitesh, 2025-06-27:
Hello @mixelpixx 

It's really nice that you created such amazing MCP server.

But when i compile i am getting errors like this

Command i run - 

`npm run build
`
The error i get

```
src/resources/board.ts:11:58 - error TS2307: Cannot find module '../utils/resource-helpers.js' or its corresponding type declarations.

11 import { createJsonResponse, createBinaryResponse } from '../utils/resource-helpers.js';
                                                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/server.ts:13:38 - error TS2307: Cannot find module './tools/project.js' or its corresponding type declarations.

13 import { registerProjectTools } from './tools/project.js';
                                        ~~~~~~~~~~~~~~~~~~~~

src/server.ts:16:38 - error TS2307: Cannot find module './tools/routing.js' or its corresponding type declarations.

16 import { registerRoutingTools } from './tools/routing.js';
                                        ~~~~~~~~~~~~~~~~~~~~

src/tools/index.ts:7:38 - error TS2307: Cannot find module './project.js' or its 
corresponding type declarations.

7 export { registerProjectTools } from './project.js';
                                       ~~~~~~~~~~~~~~

src/tools/index.ts:10:38 - error TS2307: Cannot find module './routing.js' or its corresponding type declarations.

10 export { registerRoutingTools } from './routing.js';
                                        ~~~~~~~~~~~~~~

src/tools/index.ts:13:40 - error TS2307: Cannot find module './schematic.js' or its corresponding type declarations.

13 export { registerSchematicTools } from './schematic.js';
                                          ~~~~~~~~~~~~~~~~

Found 6 errors in 3 files.

Errors  Files
     1  src/resources/board.ts:11
     2  src/server.ts:13
     3  src/tools/index.ts:7

```

Regards,
Hitesh

seraphx2, 2025-10-18:
**Update: This repo has serious fundamental issues beyond just missing source files**

I spent several hours with Claude AI investigating this repo. Here's what we found:

## Missing Source Files
The original issue is correct - critical TypeScript source files are missing from the repo:
- `src/tools/project.ts`
- `src/tools/routing.ts`
- `src/tools/schematic.ts`
- `src/utils/resource-helpers.ts`

The only reason the repo even builds is because the compiled `dist/` folder was committed (which should NEVER be in version control).

## Repository Quality Issues

**No .gitignore file**
- `dist/` folder committed (build artifacts)
- `__pycache__/` folders committed (Python cache)
- Makes you wonder how `node_modules/` didn't get uploaded too

**Git commit history**
- All commits are "Add files via upload" 
- No meaningful development history
- Suggests files were drag-and-dropped into GitHub web interface

## Critical Functional Issues

After getting it to build and connecting it to Claude Desktop, we discovered:

### 1. JSON Serialization Bug
The `load_schematic` command immediately crashes:
```
TypeError: Object of type ParsedValue is not JSON serializable
```
The Python code returns `ParsedValue` objects from the `kicad-skip` library without converting them to strings first. This causes the Python process to crash on every schematic operation.

**Location:** `python/commands/schematic.py` line 71-72

### 2. Non-Existent API Methods
The `add_schematic_component` function calls:
```python
symbol = schematic.add_symbol(...)  # This method doesn't exist!
```

The `kicad-skip` library has NO `add_symbol()` method. We tested the actual API:
- `kicad-skip` only allows **cloning** existing symbols, not creating new ones
- The `.new()` method for symbols throws `NotImplementedError: Unimplemented`
- The only way to add components is to clone an existing one with `.clone()` and modify it

**This means the advertised "schematic generation" feature is completely non-functional.**

### 3. Misleading README
The README claims:
> "This project is complete and ready for production use"
> "Successfully tested operations from project creation to component placement"
> "Add components like resistors, capacitors, and ICs to schematics"

**None of this is true.** The code was never tested. Basic operations crash immediately.

## The Fundamental Problem

**KiCAD doesn't have an official Python API for schematics** - only for PCB layout (`pcbnew`). 

This repo uses a third-party library (`kicad-skip`) that:
- Can parse and read schematic files ✓
- Can clone existing symbols (with limitations) ✓
- **Cannot create new symbols from scratch** ✗
- **Cannot access KiCAD's component libraries** ✗

The author wrote code assuming an API that doesn't exist, never tested it, and marketed it as production-ready.

## What Actually Works

Based on our testing:
- **PCB operations** (probably work, uses official `pcbnew` API)
- **Loading/reading schematics** (works after fixing the JSON bug)
- **Adding schematic components** ✗ Broken
- **Wiring components** ✗ Untested but likely broken

## The Irony

The most valuable use case (AI-assisted schematic generation) is fundamentally impossible due to KiCAD's lack of official schematic API. The PCB layout features (where the official API exists) are treated as secondary features in the README.

This repo is essentially vaporware for schematic manipulation - the main advertised feature doesn't and cannot work as described.

---

**TL;DR:** Missing source files are just the tip of the iceberg. This is an untested proof-of-concept with broken functionality, committed build artifacts, and misleading documentation. The author either used AI to generate code without testing it, or fundamentally misunderstood the libraries they were using.

TS-Hitesh, 2025-10-30:
"The man, the myth, the legend" @seraphx2 
Amazing analysis man, 

Thanks

mixelpixx, 2025-11-02:
Why and what happened: I released this on a Friday, and I found out my Dad had to have surgery,  when i showed up for that, mom announced cancer -- and both are doing great now. old, crazy, but in good health.  I apologize for this looking like abandon-ware or whatever -- but I will be checking in weekly and look forward to seeing what you think of it now that i have time to work on it